openwisp/django-netjsonconfig #21: [idea] Configuration context

nemesisdesign, 2016-02-27:
It would be very handy to have a new field JSON field that would allow to specify additional context variables (key>values) for each configuration.

This would allow to keep more logic in templates.

rohithasrk, 2017-03-04:
What do you mean by additional context variables? Can you elaborate?

nemesisdesign, 2017-03-06:
@rohithasrk the underlying library used by this app, called netjsonconfig, implements a [concept called "context"](http://netjsonconfig.openwisp.org/en/stable/general/basics.html#context).

*django-netjsonconfig* uses this context only partially: each device gets some attributes in its context, like `name`, `mac_address` or some other variables that are set globally, this feature is documented here:
https://github.com/openwisp/django-netjsonconfig#netjsonconfig-context

With this idea I thought it would be interesting to be able to have an additional JSON formatted field where an additional context can be used.

This can be useful in a situation in which that context can be automatically generated, think about automatic allocation of ip addresses.

I left this issue just as an idea, it's probably too early to start working on it. We should collect more use cases and make our mind clearer before attempting to implement it.

nemesisdesign, 2017-10-14:
[image] [[django-netjsonconfig] Device context](https://trello.com/c/glJ9oHDL/4-django-netjsonconfig-device-context)

okraits, 2018-03-19:
I might have a somewhat similar usecase:

I need to set specific mandatory settings for each device (for example DSL credentials) and I would like to have input fields for those settings appearing automatically in the device configuration for each new device.

So that the user can configure all mandatory settings easily without having to select menu items in the configuration menu, add objects, etc.

commit 1c75c6b4 (2019-01-09): [device] Add context field to device

Implements and closes #21
closed 2019-01-09 by nemesisdesign