openwisp/django-netjsonconfig #29: [Config] get rid of monkey patching in sortedm2m

nemesisdesign, 2017-02-14:
Improve the change implemented in https://github.com/openwisp/django-netjsonconfig/commit/96f4d09f0ab0fe0950fb1aa6363933057e6e7203 by supplying a patch to **django-sortedm2m**, see https://github.com/gregmuellegger/django-sortedm2m/issues/96.

rohithasrk, 2017-03-03:
Yeah. Will look into it.

nemesisdesign, 2017-03-06:
ok @rohithasrk it would be great if you could solve this. It's an interesting challenge. If you can't work on it let me know, I will have to do this soon because it's causing me to duplicate some code and I don't like this.

rohithasrk, 2017-03-06:
I see. Will try solving it ASAP.

rohithasrk, 2017-03-17:
@nemesisdesign : Made a (PR)[https://github.com/gregmuellegger/django-sortedm2m/pull/101] in django-sortedm2m. Please review it.

nemesisdesign, 2017-03-20:
@rohithasrk great, replied there

nemesisdesign, 2017-08-07:
Now we can finally upgrade the django-sortedm2m to be higher or equal to 1.5.0 and remove the monkey patching :-)

commit 54a68bdd (2017-08-08): [models] Updated sortedm2m __str__ definition #29

Closes #29
closed 2017-08-08 by nemesisdesign