openwisp/django-netjsonconfig #31: [sortedm2m] Make widget hint bolder

nemesisdesign, 2017-02-21:
The help text "Choose items and order by drag & drop." is helpful but hard to see. 

Users have suggested to make it more evident.

rohithasrk, 2017-03-02:
Working on it.

rohithasrk, 2017-03-02:
Just checked the code. Seems that it is in django-sortedm2m package and not in the django-netjsonconfig package. I understand that we need to make the text more intuitive by making it bolder. May take a while to figure out a way to edit a particular line css of a widget. Suggestions are welcome.

nemesisdesign, 2017-03-02:
@rohithasrk you can ude this file: https://github.com/openwisp/django-netjsonconfig/blob/master/django_netjsonconfig/static/django-netjsonconfig/css/admin.css

rohithasrk, 2017-03-02:
I see. Thanks. Will submit a PR sometime today. :)

rohithasrk, 2017-03-03:
Check out #32 @nemesisdesign

commit e547aac8 (2017-03-03): Merge pull request #32 from rohithasrk/hint-text-bold

Fix #31: Make the hint text bolder.
closed 2017-03-03 by nemesisdesign