openwisp/django-netjsonconfig #34: Installation package fixes

gastonche, 2017-03-04:
while trying to setup for development i encountered some problems mainly pertaining to packages that were not installed. on the migration stage i received errors about the following packages not being available or installed,

> markupsafe, functools32, cryptography

 and  from my research i found out that it's due to some dependencies having updated their own, dependencies, like **jinja** having added `markupsafe` as a dependency.
      I have added these depends to the project requirements on my fork should i make a PR for that?

rohithasrk, 2017-03-04:
This is one of the needed additions I guess. What say @nemesisdesign ?

nemesisdesign, 2017-03-06:
Thanks for reporting @gastonche.

When we find bugs in the sever side related code, we need to replicate them in the test environment.

Why these bugs are happening to you and not to me or the travis-ci build? What's different? How can I verify this is indeed a bug that needs to be addressed and not a misconfiguration?

A good way to help me verify this bug is to explain a procedure with which I can replicate the issue. Could you do that?

nemesisdesign, 2017-04-12:
I am closing this, if you find a way to replicate the issue, please re-open.

closed 2017-04-12 by nemesisdesign