openwisp/django-netjsonconfig #36: Use Vagrant to provide portable virtual environments.

rohithasrk, 2017-03-04:
There are many advantages of using Vagrant to provide portable virtual environments. It's been widely in use these days by benefiting a developer. One just needs to run a `vagrant up` command and it will isolate all the dependencies and provide a consistent environment. For more details https://www.vagrantup.com/docs/why-vagrant/ can be referred. This could perhaps be implemented in all the OpenWISP projects. Suggestions please.

nemesisdesign, 2017-03-06:
Yes it could be interesting. What needs to be implemented exactly? A vagrant file?

rohithasrk, 2017-03-06:
Yes. A Vagrant file is all that is needed.

nemesisdesign, 2017-03-06:
A vagrant file defaulting either to Ubuntu 16 LTS is probably a good default, plus a few lines in the README to explain how to use it.

rohithasrk, 2017-03-06:
Yes. Shall I take this up?

nemesisdesign, 2017-03-06:
#29 would be very helpful to me and is more related to django. This one would be useful to new comers but it's not very django-related. Up to you, depending on which time you prefer and how much skill you have.

rohithasrk, 2017-03-06:
Yes. Working on #29 first.

nemesisdesign, 2017-09-08:
Closing for inactivity.

closed 2017-09-08 by nemesisdesign

nemesisdesign, 2018-08-27:
https://github.com/openwisp/vagrant-openwisp2 (thanks to @hispanico)