openwisp/django-netjsonconfig #44: [UI] configuration variables clash with validation

nemesisdesign, 2017-03-13:
The following NetJSON:

```javascript
{
    "interfaces": [
        {
            "type": "virtual",
            "name": "tap0",
            "mac": "{{ mac_address }}"
        }
    ]
}
```

Won't validate. This sucks big time! 

We should find a solution for this problem. Maybe we can evaluate the variables with javascript. I don't know.

yasharmaster, 2017-03-14:
I'm working on this.

nemesisdesign, 2017-03-14:
@yasharmaster let us know how you intend to solve it once you get an idea

saksham1115, 2017-03-21:
How can I recreate this validation ?

adriankirchner, 2018-04-22:
@yasharmaster @nemesisdesign are there any updates on this? If I get that right, atm it is not possible to eg. specify the lan ipv4 address via a custom variable.
Therefore I can't use a template to configure the eth0 interface but need to add this configuration to every device manually
[image]

nemesisdesign, 2018-04-23:
@adriankirchner as far as I know, nobody is currently actively working on this issue

nemesisdesign, 2018-10-25:
@adriankirchner solving this will require some work, I've created #93 to summarize the different actions required to fix it.

nemesisdesign, 2020-06-30:
Fixed by https://github.com/openwisp/django-netjsonconfig/pull/178

closed 2020-06-30 by nemesisdesign