openwisp/django-netjsonconfig #45: [UI] Improve advanced editor widget

nemesisdesign, 2017-03-24:
@gastonche: there are few things I noticed after merging #43 

- hitting back after opening the advanced mode on the browser produces inconsistent results
- the label with netjsonconfig-hint is duplicated, but now we have 2 elements with the same ID in the page, and this is bad
- when an invalid json value is entered and the exit button is clicked we get the advanced editor but not in full screen

gastonche, 2017-03-27:
@nemesisdesign I have been occupied for the last 2 days, so I have been quite absent. so did you fix all the issues here with this merge, or is there something I may have to work on here again?

nemesisdesign, 2017-03-27:
@gastonche, the first 3 items of the list in the issue description are unresolved.
Also try to take a look at the improvements I added to your patch: https://github.com/openwisp/django-netjsonconfig/commit/0dc39c4ba555333a75b062f8a4363ede37d3a181

nemesisdesign, 2017-03-28:
Closed by #46

closed 2017-03-28 by nemesisdesign