openwisp/django-netjsonconfig #47: [idea] filter configurations by relationship with a specific template

cappe87, 2017-03-29:
it may be useful to have a way to filter configurations that have a relationship with a specific template

nemesisdesign, 2017-03-29:
@cappe87 adding the filter is easier than I thought. It's just a matter of adding `templates`in `list_filters` of `AbstractConfigAdmin`:
https://github.com/openwisp/django-netjsonconfig/blob/master/django_netjsonconfig/base/admin.py#L214

The hard part is in [openwisp-controller](https://github.com/openwisp/openwisp-controller): the filter should display only templates of organizations that the current user has access to (only superusers see everything).

nemesisdesign, 2017-03-29:
I created an issue also on https://github.com/openwisp/openwisp-controller/issues/2.

nemesisdesign, 2017-04-12:
Closed by 90d4e99

closed 2017-04-12 by nemesisdesign