openwisp/django-netjsonconfig #49: [UI] full screen covers preview

nemesisdesign, 2017-03-31:
Full screen mode covers preview config mode.

It's probably just a matter of adjusting `z-index` css property so the ones of the preview is higher.

@gastonche would you like to address this?

gastonche, 2017-04-01:
I'll Address that right away @nemesisdesign

gastonche, 2017-04-01:
@nemesisdesign, Can't reproduce the full screen covering preview. Please directions

nemesisdesign, 2017-04-01:
@gastonche right, sorry I didn't explain you everything.

There's a shortcut for preview mode, which is ALT+P. If you use this shortcut while in full screen mode you will notice nothing happens. Try then to close the full screen and you will see the preview.

nemesisdesign, 2017-04-11:
thx for fixing this @gastonche

closed 2017-04-11 by nemesisdesign