openwisp/django-netjsonconfig #55: [config-editor] VPN servers JSON-schema validation fails

nemesisdesign, 2017-05-04:
How to reproduce:

- go to VPN servers
- add new
- fill in the minimum required information, click save and continue
- now try to go into advanced mode, notice the validation failures, even though the entered JSON is considered valid by other json validation libraries

gastonche, 2017-05-08:
working on this now

nemesisdesign, 2017-05-08:
Great, keep us up to date 👍

nemesisdesign, 2017-05-10:
@gastonche have you been able to reproduce the issue?

I thought that it happens because the validation library used by the json-schema editor and the advanced mode editor is different. The one used by json-schema editor works correctly.

gastonche, 2017-05-10:
Yeah I thought about that too, I plan on finding a way to modify the
validation library in the Advanced JSON Editor

nemesisdesign, 2017-05-10:
@gastonche if we could configure the advanced editor to use the same JSON-schema validation mechanism of the JSON-Schema UI generator it would be perfect

gastonche, 2017-05-10:
Yeah I am trying to do research on that currently.

nemesisdesign, 2017-05-18:
@gastonche any update on this?

gastonche, 2017-05-18:
Hello Guys,
          I have been working on this for the past two days and have been
getting very funny outcomes. I have tried to use a  custom instance of ajv
with the advanced mode json editor to no avail.
          The strange thing with my tests is the fact that  when I use an
instance of the jsonEditor with out of the django-netjsonconfig with the
same configs as with the configs as within the django-netjsonconfig, it
does validate correctly. SO right now I don't understand what is wrong. But
At least now I know the problem is not with the advanced editor itself. I
am using the same files for the test instance as with the netjsonconfig so
I am positive that it was supposed to produce the same result.
          I would like some ideas if any on what may be the problem here.
 however i continue trying other tests to find out what is the difference.

nemesisdesign, 2017-05-19:
@gastonche Add a way to easily disable the schema validation. I will disable this feature in the VPN management form.

gastonche, 2017-05-19:
Okay, will get that done immediately

commit 94e158ae (2017-05-24): [config-editor] Disabled schema validation for VPN #55

Closes #55
closed 2017-05-24 by nemesisdesign