openwisp/django-netjsonconfig #62: Add hostname validator to Device.name

nemesisdesign, 2017-10-19:
How to reproduce the issue: try to save a new device with name `test device`.
Result: netjsonconfig validation complains test device is not a valid hostname and the validation error is shown in the configuration section.
Fix: it would be better to show this validation error relating directly to the name field

R9295, 2017-12-14:
@nemesisdesign I was able to create a new device with the name set to `test device`. I didn't receive a validation error.

R9295, 2017-12-14:
[image]

nemesisdesign, 2017-12-14:
try creating a new device and also a configuration (scroll down, click "add configuration")
the error is triggered by the configuration, which is a oneToOne optional relation between Device and Config

commit 31d04aac (2017-12-20): [models] Added a name validator to Device.name #62

Implements and closes #62
closed 2017-12-20 by nemesisdesign