openwisp/django-netjsonconfig #63: JSON validation error

fabricat, 2017-10-27:
While editing a template in the openwisp-controller, after having switched to "Advanced mode (raw JSON)" i see these errors:
[image]
The error disappears when I add any alphabetical character at the beginning of an hostname (see the active line above).

nemesisdesign, 2017-10-27:
@fabricat thanks for moving the issue, does the problem happen anyway if you remove the "openvpn" section?

nemesisdesign, 2017-11-06:
@fabricat does the problem happen anyway if you remove the "openvpn" section?

fabricat, 2017-11-06:
Sorry, I forgot to test that.
However removing the "openvpn" section doesn't change the result.

Instead I tried adding a digit at the beginning of the .openvpn[0].remote[0].host property and... the same error appears also here.

It looks like the problem is related to the pattern validation of any hostname beginning with a digit.

nemesisdesign, 2017-11-06:
@fabricat ok thx

okraits, 2018-08-09:
Also stumbled over this bug...

nemesisdesign, 2018-10-25:
I forgot to reply here!
I think this is due to the jsonschema validation library we use, we should check if using the latest version of it fixes this issue (we have an issue for trying the latest version of jsonschema in the netjsonconfig library https://github.com/openwisp/netjsonconfig/issues/114).
We'll need a unit test for this case.
We could try to get some help from some of the Google Code-In studens here.

nemesisdesign, 2020-05-22:
The issue is caused by https://github.com/openwisp/netjsonconfig/issues/132.

There is a proposed solution there but the UI in this module needs to be updated as well (CSS).