openwisp/django-netjsonconfig #64: [rare] Autoregistration fails but device is created anyway

nemesisdesign, 2017-11-07:
In rare cases, auto-registration may fail because of misconfiguration or bugs introduced during development.

It then may happen that even if the auto-registration fails, the device may be created anyway, eg: if the uncaught exception is triggered in the phase which adds templates (m2m relationship in django are saved after the main object has been created).

This is not good because it triggers other unexpected behaviour, a chain reaction which is hard to debug.
The ideal solution would be to ensure to catch any potential exception during all the critical phases of the autoregistration and if any of these phases fails, the system should log the error, trigger something like a 400 bad request and delete the newly created invalid device.

pacicap, 2019-02-28:
i am interested in solving this issue please may i have some tips on how to get started?

pacicap, 2019-03-01:
please on which file should i concentrate the most i am not very familiar with javascript

atb00ker, 2019-12-30:
@pacicap Hey please join the gitter channel to ask questions, sometimes the questions here get lost in the noise! :smile:

commit 3d143343 (2020-01-07): [auto-registration] Improved PR #139

Fixes #64
Closes #139
commit e7d70a04 (2020-01-07): [auto registration] Ensure atomicity of transactions with database #64

Wrapped auto-registration database updates with atomic blocks to avoid inconsistent states. Fixes #64
commit 849f9cde (2020-04-14): [auto registration] Ensure atomicity of transactions with database #64

Wrapped auto-registration database updates with atomic blocks to avoid inconsistent states. Fixes #64
commit c64c6a50 (2020-04-14): [auto-registration] Improved PR #139

Fixes #64
Closes #139
closed 2020-01-07 by nemesisdesign