openwisp/django-netjsonconfig #77: [bug] If default templates contain a flaw, auto-registration will fail

nemesisdesign, 2018-04-07:
If the default templates have a conflict with one another which causes the resulting NetJSON configuration to fail the schema validation, auto-registration will fail with 500 internal server error.

A possible solution would be the following:

- if the current template being saved is flagged as default
- before saving, create a NetJSON configuration from all the default templates using the same backend and try to validate it against the schema used, if the validation fails, trigger a django `ValidationError`

We must provide a way to extend this behaviour, since downstream apps will need more complex validation rules (openwisp-controller will have to check default templates for each organization if the default template being edited is "shared").

oltarzewskik, 2019-07-02:
@nemesisdesign I am not able to reproduce this bug, I was doing following:
I created two `templates` `Enabled by default` with various timezones.
I created device configuration with these `templates` enabled by default, but I haven't seen any errors in my browser's console and in my terminal where is running django server. While watching preview of configuration before saving and after saving I only saw that one `template` is overriding another one in both cases.

I was using chromium browser for it.

nemesisdesign, 2019-07-18:
Creating two templates with different timezones will not reproduce this error.

Try creating two default template with an interface which is exactly identical and will trigger a schema error.

oltarzewskik, 2019-08-10:
@nemesisdesign I followed your tips, but still I don't see any error. Can you check is that error still occuring on the latest version of `django-netjsonconfig`?

atb00ker, 2020-04-12:
Reopened based on comment here: https://github.com/openwisp/django-netjsonconfig/commit/ffd53171c8db3f98559d934320e52575dd98da09#commitcomment-38440339

nemesisdesign, 2020-04-13:
Thanks @atb00ker and @nepython! :+1:

commit ffd53171 (2020-04-11): [signals] Improved PR #77

Closes #77