openwisp/django-netjsonconfig #86: [feature] Hardware id field

nemesisdesign, 2018-10-03:
Follow up to mailing list discussion: https://groups.google.com/d/msg/openwisp/CGJKYfIregU/yd3-ma9GAAAJ

We should add a dedicated field in django-netjsonconfig with a generic name which can be used for any type of hardware ID. Since it's generic, we should be flexible in defining this field, but we know for sure it's going to be a unique alphanumeric string with a generous length, the database field can be NULL (and if an empty string is supplied it should be converted to None to avoid triggering the unique DB constraint error).

The field may or may not be required depending on a configuration (django setting), which makes it not required by default, which maintains backward compatibility.

The human readable label and the help text of the field should be configurable, by default we could use something like "Serial number", but it can be modified to use a different term depending on the context where it is used.

In my opinion this feature should be deactivated by by default and shown only when a specific setting is turned on.

When the feature is enabled, the following things should happen:

- it should be possible to search devices in the admin by their serial number
- it should be shown in the device list, probably as first field from the left
- it should be shown and be editable in the add device page, shown right before or right after the name field 
- it should be shown and (probably) readonly in change device page, shown right before or right after the name field
- the controller view which allows registration of new devices should save this value if it finds it in the HTTP POST data (this may already work now due to how that part is implemented)
- the field should become available in the device context

okraits, 2018-10-26:
When trying to make the hardware ID/serial number field readonly only for the change device page, I noticed that the following fields are not readonly in the change device page:
* Mac address
* Key

I would think that the values of these fields should not change for a specific device.
On the other hand, there's the recurring usecase that a device is defective and needs to be replaced by a new one. Then it would be handy to use the existing device entry and just update the relevant fields, like serial number and mac address.

Thus I would suggest to not make the serial number field readonly.
@nemesisdesign What do you think?

nemesisdesign, 2018-10-26:
@okraits makes sense allowing to change it, it's more flexible. In the future we may make a widget that shows it as readonly but allows changing it with a button if needed. 

Regarding the device context I mean:

- https://github.com/openwisp/django-netjsonconfig/blob/master/django_netjsonconfig/base/config.py#L87-L100
- which is documented in the README: https://github.com/openwisp/django-netjsonconfig#netjsonconfig-context

PR #95 (2018-11-07): WIP: [feature] Add support for hardware ID / serial number
Closes #86
commit b538c721 (2018-11-07): [hardware-id] Fixed tests #95 #86

Closes #95
Closes #86
closed 2018-11-07 by nemesisdesign