openwisp/django-netjsonconfig #91: TypeError: can only concatenate list (not "str") to list

imp1sh, 2018-10-10:
Hi,

I'm really new to openwisp / netjsonconfig.
I made a backup of two different systems (Archer C7 v2 and X86_64 Qemu guest).
Both I tried to reverse with the following lines:
```
from netjsonconfig import OpenWrt
#router = OpenWrt(native=open('./backup-g0.sim.libcom.de-2018-10-10.tar.gz'))
router = OpenWrt(native=open('./backup-OpenWrt-2018-10-10.tar.gz'))
print(router.render)
```
In both cases this results in the following error:
```
 python reverse.py 
Traceback (most recent call last):
  File "reverse.py", line 3, in <module>
    router = OpenWrt(native=open('./backup-OpenWrt-2018-10-10.tar.gz'))
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/base/backend.py", line 47, in __init__
    self.parse(native)
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/base/backend.py", line 280, in parse
    self.to_netjson()
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/base/backend.py", line 293, in to_netjson
    value = converter.to_netjson()
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/base/converter.py", line 108, in to_netjson
    result = self.to_netjson_loop(block, result, index + 1)
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/openwrt/converters/interfaces.py", line 206, in to_netjson_loop
    interface = self.__netjson_interface(block)
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/openwrt/converters/interfaces.py", line 217, in __netjson_interface
    interface = self.__netjson_addresses(interface)
  File "/usr/lib/python2.7/site-packages/netjsonconfig/backends/openwrt/converters/interfaces.py", line 262, in __netjson_addresses
    for address in ipv4 + ipv6:
TypeError: can only concatenate list (not "str") to list
```

imp1sh, 2018-10-11:
I'm sorry I might have posted this in the wrong repo, i.e. django-netjsonconfig instead of just netjsonconfig...

nemesisdesign, 2018-10-12:
@imp1sh yes, that's https://github.com/openwisp/netjsonconfig, looks like there's a similar issue: https://github.com/openwisp/netjsonconfig/issues/104. Could you please move the issue in the right repo to help us track it better?

imp1sh, 2018-10-13:
I created issue in correct repo now: https://github.com/openwisp/netjsonconfig/issues/118

closed 2018-10-13 by imp1sh