openwisp/django-netjsonconfig #93: [variables] Rethink validation to allow using variables in templates

nemesisdesign, 2018-10-25:
This issue is related to:

- #92 
- #44

The goal of this issue is to make the following use cases work without errros:

- add a template (both programmatically or from the web UI) which defines an interface which has an IP defined in a variable
- create a device (both programmatically or from the web UI) which uses the template just created in the previous step and define the variable in its context (needs #92)

What should happen:

- the template should be created without validation errors
- the device should be created without validation errors if the variable in the device context is defined correctly
- a validation error should be raised if
  - the variable is missing from the device context
  - the variable is defined but its format is invalid, eg: if we are expecting an ip address but we define a string which is not a valid ip address

This is not trivial and will likely require intervention in different modules (especially [netjsonconfig](https://github.com/openwisp/netjsonconfig/)), but it's an important feature OpenWISP should have had since the beginning but it doesn't still have because of these bugs.

NoumbissiValere, 2019-05-26:
@nemesisdesign please can you direct me to the part of the project where validation for variables in templates is carried out?
Thanks

NoumbissiValere, 2019-05-27:
I followed through the execution of the code right up till validation. And i noticed that writing `{{ mac_address }}` doesn't validate because of the schema.
The schema is good, it is perfect
But it has a fixed pattern which is the pattern for a valid mac address and its good. But now i noticed `{{ mac_address }}` was against the pattern that was specified. reason why the validation failed.
Thus i tried adding a pattern to the existing one which will validate `{{ some character}}` as valid. with this, the validation processed passed both for when there is actually a valid address and  `{{some character}}`
after succeeding in creating this template with mac address as a variable, I went to the device add and used the template i created and specified the value for the variable in the context field. I tested the values on the openwrt VM and it work.
I did the same approach for IP address and it worked.
So i think our variable validation problem is at the level of the pattern that was specified in the schema for the different fields. to enable variable validation both in the web UI and advanced mode, we have to add the pattern to validate variables to the existing patterns.

NoumbissiValere, 2019-05-27:
Also, I feel that even if we validate this with javascript or at the `clean` method, when `backend.validate()` is executed, the validation will still fail because `validate()` validates `config` against the schema and the pattern in the schema will reject the values entered since they don't match the pattern specified.

nemesisdesign, 2019-05-28:
I don't understand why are you mentioning to change the schema,the schema is not broken. The validation flow is broken. I sent you some messages on gitter.

nemesisdesign, 2020-06-30:
This was fixed by https://github.com/openwisp/django-netjsonconfig/pull/178

commit 78380e62 (2019-06-12): [validation] Frontend variable validation #93

Modified the widget.js to ensure the variables
are replaced by their default values before
validation is called.

Fixes #93
commit 9285773e (2019-06-12): [Variable] Use default values in config #93

This PR enures that default valuaes
set at template creation is set to
Config context to be used as default
values for variables if none is supplied
at the context field

Fixes #93
closed 2020-06-30 by nemesisdesign