openwisp/django-netjsonconfig #104: [vpn] Add controller views (download & checksum) for VPN config

nemesisdesign, 2019-01-17:
Some users want to have the VPN server to automatically updated when the server config changes on OpenWISP, but this is not doable right now.

To achieve this we should proceed with something like the following:

- add a key field to `Vpn` model (as Device has) + tests
- add a `checksum` view for VPN objects (like it's done for devices) + tests
- add a `download` view for VPN objects (like it's done for devices) + tests
- update the README to mention this feature and how to use it (eg: show a bash script example to run in cron on debian)

oltarzewskik, 2019-01-20:
@nemesisdesign I'd like to create PR to first dot, my question is: have I to generate key value (like in the device) or let user to create own? (I assume that I have to generate key, but I want be sure)

nemesisdesign, 2019-01-21:
@oltarzewskik generate, but you shouldn't write the logic from scratch, because most of that logic is already present in `django-netjsonconfig`, read the code and try to reuse as much as possible

oltarzewskik, 2019-01-21:
@nemesisdesign Got it :+1:

PR #105 (2019-04-23): [WIP] [vpn] Add controller views (download & checksum) for VPN config #104
- [x] add a key field to `Vpn` model (as Device has) + tests

- [x] add a `checksum` view for VPN objects (like it's done for devices) + tests

- [x] add a `download` view for VPN objects (like it's done for devices) + tests

- [ ] update the README to mention this feature and how to use it (eg: show a bash script example to run in cron on debian)

Fixes #104
commit 3d0d7588 (2019-04-23): [vpn] Add controller views (download & checksum) for VPN config #104

Fixes #104
closed 2019-04-23 by nemesisdesign