openwisp/django-netjsonconfig #107: [vpn] DH params in preview are exported as python bytes repr

nemesisdesign, 2019-03-05:
Steps to reproduce:

- create a VPN server object
- hit preview

Notice something like the following in the preview:

```
# path: dh.pem
# mode: 0644

b'-----BEGIN DH PARAMETERS-----
MIGHAoGBAKQku25Xa+mkCz264wbefgyFIt3ir1vR1giDcx9LvK4ml7mNywD01HaK
L6X8aYY4hmZN4B0Sj/quTNKx0BaFLK6nhCH2LIZl2FaR9KV1i7caXM4mHNn6gCID
WtuSpjETCa7TRHF+AYOeOAc00vg3+1dAxKNYx4M8IwqNKxCES5prAgEC
-----END DH PARAMETERS-----
'
```

The correct output would be:

```
# path: dh.pem
# mode: 0644

-----BEGIN DH PARAMETERS-----
MIGHAoGBAKQku25Xa+mkCz264wbefgyFIt3ir1vR1giDcx9LvK4ml7mNywD01HaK
L6X8aYY4hmZN4B0Sj/quTNKx0BaFLK6nhCH2LIZl2FaR9KV1i7caXM4mHNn6gCID
WtuSpjETCa7TRHF+AYOeOAc00vg3+1dAxKNYx4M8IwqNKxCES5prAgEC
-----END DH PARAMETERS-----

```

atb00ker, 2019-03-07:
Hi,

Can you please help me reproduce this?
I thought it's a simply python compatibility issue.
However, I did not face it on the master branch, tested it with `python2.7.15rc1` and `python3.6.7`.
Also tested it by installing from the ansible-role on Ubuntu18.04 and still didn't face it.

Thanks

nemesisdesign, 2019-05-01:
@atb00ker sorry Ajay for not replying to your question, I'm not sure why you could not replicate it. I would have tried with python 3.5 and latest version of all the dependencies but it should be fixed now.

PR #110 (2019-05-01): [vpn] Fixed DH params in preview and download configuration #107
Fixes #107
commit 9730a0b4 (2019-05-01): [vpn] Fixed DH params in preview #107

Fixes #107
closed 2019-05-01 by nemesisdesign