openwisp/django-netjsonconfig #108: [django] Add support for django 2.2

nemesisdesign, 2019-04-02:
Add support for django 2.2

atmishra, 2019-09-02:
I would like to work on this. Can I take this?

NoumbissiValere, 2019-09-02:
@atmishra it looks like @EdgeKing810  has solved the issue. 
Please can you look for another issue?

IAmAMissingColon, 2019-12-11:
Hello I think this issue has been solved in #132 as well

nemesisdesign, 2019-12-11:
@IAmAMissingColon we need to add support for django 2.2 now and release before we can merge #132, if you want you can help with this.

IAmAMissingColon, 2019-12-11:
Yeah, I already claimed the task :smile:

IAmAMissingColon, 2019-12-12:
Hello !

* Pytz 
* jsonschema
* jinja2
* OpenSSL

the above dependencies needs to be installed/upgraded while enabling support for Django2.2

any suggestion ?

IAmAMissingColon, 2019-12-12:
errors I faced:

1. error: `django-model-utils 4.0.0 is installed but django-model-utils<3.3.0,>=3.1.2 is required by {'openwisp-utils'}`
2. `django-x509 0.5 requires cryptography<2.9.0,>=2.4.0, which is not installed.`
3. `django-x509 0.5 requires pyopenssl<20.0.0,>=17.5.0, which is not installed.`

after running `./manage.py migrate` 

1. ModuleNotFoundError: `No module named 'jsonschema'`

after installing jsonschema (pip install jsonschema)

2. ModuleNotFoundError: `No module named 'jinja2'(pip install jinja2)`

after installing jinja2

3. ModuleNotFoundError: `No module named 'OpenSSL'`

after installing OpenSSL (pip install pyOpenSSL)

./manage.py migrate run without giving any error. 
./manage.py runserver runs without any error.

but unfortunately tests fails same as @EdgeKing810

nemesisdesign, 2019-12-12:
It is stated in the issue description:

> Some automated tests fail with django 2.2 so it will be necessary to do some work there.

You have to research and find out why those tests are failing and fix them on django 2.2.

PR #134 (2019-12-20): [requirements] Django2.2 #108
Enabled Django2.2

Fixes #108
commit a55b881d (2019-12-20): [requirements] Django2.2 #108

Enabled support for Django 2.2 and fixed JS bug introduced in  Django2.2

Fixes #108
commit d51f697b (2020-04-14): [requirements] Django2.2 #108

Enabled support for Django 2.2 and fixed JS bug introduced in  Django2.2

Fixes #108
closed 2019-12-20 by nemesisdesign