openwisp/django-netjsonconfig #111: [bug] Unable to add new device

oltarzewskik, 2019-05-01:
After merging [this](https://github.com/openwisp/django-netjsonconfig/commit/69289cff5458391477460221e28e7b6559381c73) commit I am unable to add new device by web UI
[image]

nemesisdesign, 2019-05-02:
Have you tried clearing the cache of the browser?

What does appear in the javascript console?

nemesisdesign, 2019-05-02:
ensure also openwisp-controller and openwisp-utils are updated to the latest dev version in your env

atb00ker, 2019-05-02:
> Have you tried clearing the cache of the browser?

Yes.

> What does appear in the javascript console?

```
Uncaught TypeError: Cannot read property 'match' of undefined
    at prepareUl (widget.js:27)
    at HTMLUListElement.<anonymous> (widget.js:53)
    at Function.each (jquery.min.js:2)
    at w.fn.init.each (jquery.min.js:2)
    at iterateUl (widget.js:50)
    at HTMLDocument.<anonymous> (widget.js:60)
    at l (jquery.min.js:2)
    at c (jquery.min.js:2)
```
> ensure also openwisp-controller and openwisp-utils are updated to the latest dev version in your env

I'll crosscheck and get back about it asap! :smile:

nemesisdesign, 2019-05-03:
Thanks @atb00ker for the info.

If I could replicate this bug I think I would be able to fix it quickly.

Where does that error come from? Which JS file is failing?

atb00ker, 2019-05-03:
The error is seems to be a django jquery update related problem.
On django-2.1.8: the error in is: `sortedm2m/widget.js`.

Here is my requirements.txt:
``` 
asgi_redis 
service_identity 
django-redis
psycopg2
-e git+git://github.com/openwisp/openwisp-controller#egg=openwisp_controller 
https://github.com/openwisp/django-netjsonconfig/tarball/master
https://github.com/openwisp/django-x509/tarball/master
-e git+git://github.com/openwisp/openwisp-users#egg=openwisp_users
-e git+git://github.com/openwisp/openwisp-utils#egg=openwisp_utils
uwsgi
```

You can also use the docker-compose [here](https://github.com/atb00ker/dockerize-openwisp) to relicate the error since the controller container is affected on it as well! 
I'll try to look further into solving it asap! :smile:

atb00ker, 2019-05-03:
Okay a little bit more information, the line 27 in `sortedm2m/widget.js` is failing which is `label.attr('for').match(/^(.*)_\d+$/)[1]` because label is:
```
<ul class="sortedm2m-items hide sortedm2m">
</ul>
```
and it does not have `attr('for')`
I am able to replicate it now simply by doing:

```
git clone https://github.com/openwisp/openwisp-controller 'tcontroller' && cd tcontroller;
pipenv install . --skip-lock --dev --three
pipenv shell
./install-dev.sh
```
Now, migrating and running the server should give the error!

atb00ker, 2019-05-05:
**Important update:** The `sortedm2m/widget.js` issue seems irrelevant to this issue.
I tried an instance of a previous version of openwisp2 in which `devices form` is working perfectly. The `sortedm2m/widgets.js` error is on that version as well. 
Hence, the information i've send above is not relevant to this issue.

atb00ker, 2019-05-05:
Okay, I found it!

Adding this CSS displays the hidden form:
```
.change-form #device_form > div > fieldset.module.aligned {
    display: block;
}
```

@nemesisdesign, [this](https://github.com/openwisp/django-netjsonconfig/commit/69289cff5458391477460221e28e7b6559381c73#diff-4c4b4d234f633e7155601c3c5bea108bR166) line in `admin.css` was setting it to `display: none` it was changed 22 days ago in [this commit](https://github.com/openwisp/django-netjsonconfig/commit/69289cff5458391477460221e28e7b6559381c73).

nemesisdesign, 2019-05-22:
@atb00ker can you clear your cache and try again please?
EG: open the JS console of chrome and check "Disable cache", then reload the page.

atb00ker, 2019-05-23:
@nemesisdesign Yes, the issue presists. I have tried:
- No cache reload
- Hard reloads (reloading without any previous data)
- Opening in incognito
- Opening in another browser.

Hence, I am inclined to believe that the issue is not due to cache data.

nemesisdesign, 2019-05-23:
I have been to replicate it.

nemesisdesign, 2019-05-24:
Should be fixed, thanks guys! :+1:

commit f99cd1fa (2019-05-24): [admin] Fixed add form CSS #111

Fixes #111
closed 2019-05-24 by nemesisdesign