openwisp/django-netjsonconfig #116: [validation] JavaScript validation fails.

NoumbissiValere, 2019-06-04:
with respect to [PR 115](https://github.com/openwisp/django-netjsonconfig/pull/115/commits/78380e62d8c942ec84ce5605ad88b1687dad00af), valid default values for variables can be validated with the schema backend. Even though the schema validates the data, the JavaScript don't still validated the data entered. It shows input error.
**How to reproduce the bug**
 do the following in advance mode and hit the exit button, then look at the error message at the `mac_address` field.
{
    "interfaces": [
        {
            "type": "ethernet",
            "disabled": false,
            "mac": "{{ mac_address }}",
            "mtu": 1500,
            "addresses": [],
            "name": "eth0",
            "autostart": true,
            "network": ""
        }
    ]
}
with these defaults:

{
    "mac_address": "00-11-22-33-44-56"
}

okraits, 2019-06-19:
:+1: @NoumbissiValere Good job! As I'm using the device context to be able to use device-specific data in templates, I'm looking forward to this as well. I will possibly test it in the next days.

nemesisdesign, 2020-06-30:
Closed by https://github.com/openwisp/django-netjsonconfig/pull/178

PR #117 (2019-06-28): Issue/116
[validation] JavaScript validation fails #116

Replace the variables with default values
before widgets validation in order to make
the test pass with valid default values.

Fixes #116
commit 0828a652 (2019-06-28): [validation] JavaScript validation fails #116

Replace the variables with default values
before widgets validation in order to make
the test pass with valid default values.

Fixes #116
closed 2020-06-30 by nemesisdesign