openwisp/django-netjsonconfig #131: [ feature ] Allow more actions involving configuration templates

joaohcca, 2019-12-03:
Admins should be able to create templates based on existing templates.
 
Additionally, multi-organization admins should be able to import/clone templates between orgs.

atb00ker, 2019-12-05:
An easy non-GUI approach to this would be copies the json from advanced-json mode and pasting it! :smile:

nemesisdesign, 2019-12-09:
Here's a solution that should work.

We need to add a new admin action in the template admin, that allows to clone selected templates.

We need an [intermediate page](https://docs.djangoproject.com/en/2.2/ref/contrib/admin/actions/#actions-that-provide-intermediate-pages) in which the user selects the organization (if they have access to more than one organization).
The intermediate page should show a message asking the user in which organization they want to clone the templates, making sure only organizations to which user has access to are shown.
 
When the templates are cloned, the template name should be changed to `(COPY) <name>`, where `<name> is the original template name.

pawelplsi, 2019-12-15:
In my opinion it would be better to implement this feature in openwisp-controller repository instead of this one, because it would violate the modularity of project if we implemented organization selection intermediate page in django-netjsonconfig repository which has nothing to do with oranizations/multitenancy. Am I right?

atb00ker, 2019-12-16:
@pawelplsi Multitenancy will not be moved to django-netjsonconfig instead the template clone part would be done in django-netjsonconfig and multi-org part would be done seperately.
For that I think we need to make 2 different pull requests. I think it'll be best if you first do the part related to django-netjsonconfig.Add a template cloning feature based on we will move ahead with the openwisp-controller part, what do you think?

nemesisdesign, 2020-01-09:
Implemented in #140.

closed 2020-01-09 by nemesisdesign