openwisp/django-netjsonconfig #159: [enhancement] Duplication of action list, save row

nepython, 2020-03-09:
**Duplicate display of action list**
[image]

**Duplicate display of submit row**
[image]

nepython, 2020-03-09:
I traced the cause of it and it seemed as if it has been done on purpose, so @nemesisdesign can you please confirm if these are required or else I move ahead and fix it. :grinning:

atb00ker, 2020-03-09:
Yes this is on purpose, all django admin pages are designed like this. These sections are on the top and the bottom of the page! :smile:

closed 2020-03-09 by atb00ker

nemesisdesign, 2020-03-09:
It's very useful when the system is fully working and there's some content in it. 
It looks weird when the system is empty.

nepython, 2020-03-09:
[image]
Though I think it can be removed from some places like the one above.

nemesisdesign, 2020-03-10:
Not a priority right now, focus on the priorities