openwisp/django-netjsonconfig #165: [admin] 404 on Device change page raises 500 internal server error

nemesisdesign, 2020-03-13:
How to replicate:

- open device admin list
- open a device change page with the UUID modified so to trigger a 404

Expected result: 404

Actual result: 500 internal server error.

A test to ensure we won't have regressions will be needed.

nepython, 2020-03-17:
> A test to ensure we won't have regressions will be needed.

@nemesisdesign, is the test supposed to ensure that we get a **404** error when trying to access a page with a URL containing the wrong uuid, while we get a status code of **200** when trying to access the same page with uuids present in the database `or` I am also supposed to ensure that I don't break anything or reintroduce old bugs (If so is the case, can you please guide me to those bugs).

nemesisdesign, 2020-03-17:
@nepython write a test which fails with the bugged version of the code (before you apply the fix), then apply the fix and ensure the test passes. The test should expect 404, there should be [similar tests](https://github.com/openwisp/django-netjsonconfig/blob/ce410f8cca0936c44004772f277d5fee2d5a3f26/django_netjsonconfig/tests/test_admin.py#L97-L101) already so you just need to adapt one.

PR #167 (2020-03-22): [admin] 404 on Device change page raises 500 internal server error
close #165
commit 01d334bc (2020-03-22): [admin] Fixed 500 in change device #165

Closes #165
closed 2020-03-22 by nemesisdesign