openwisp/django-netjsonconfig #169: [x509] VPN client files should be created with 0600 permissions by default.

nemesisdesign, 2020-03-21:
VPN client files (ca, cert, key) should be created with 0600 permissions by default.

TheOneAboveAllTitan, 2020-03-21:
Working on this!

TheOneAboveAllTitan, 2020-03-21:
@nemesisdesign it is kinda hardcoded [in netjsonconfig](https://github.com/openwisp/netjsonconfig/blob/75a77efdc9782b0d16b910b82f2d7f58951da232/netjsonconfig/backends/openvpn/openvpn.py#L124). One way can be to make DEFAULT_FILE_MODE a class variable, whose value is updated from schema.

nemesisdesign, 2020-03-22:
@TheOneAboveAllTitan thanks for the analysis, I think this is the way to go:
https://github.com/openwisp/netjsonconfig/issues/147#issuecomment-602113886

The problem is downstream. The test in this module will need to be updated like you did in https://github.com/TheOneAboveAllTitan/django-netjsonconfig/commit/7bb6c8177622b923d586797daa358660f9753cb1.

nemesisdesign, 2020-03-28:
So now that https://github.com/openwisp/netjsonconfig/pull/148 is merged, here it remains pending to verify that the fix works when used via the admin and then we can close this as well.

PR #171 (2020-04-05): [x509] VPN files have 0600 permissions by default #169
Set default permission of files to 0600 and modified tests accordingly.
Minor fix in tests/settings.py
Closes #169
Applies changes from https://github.com/openwisp/netjsonconfig/pull/148/

[image]
commit 4e084adf (2020-04-05): [x509] VPN files have 0600 permissions by default #169

Set default permission of VPN files to 0600 and modified tests accordingly.
Closes #169
closed 2020-04-05 by nemesisdesign